hytech-racing/MCU #39: Create multiple ticking functions for MCUInterface

walkermburns, 2024-02-10:
[image]

The group of functions called in the MCUInterface tick is very diverse, and it should be split up into multiple tick functions with different frequencies.  For example, MCU status messages do not need to be sent at a high frequency, around 10Hz, but the read_mcu_status() function that stores the current state of the shutdown read pins is integral to the functioning of the safety system and state machine and should be ticked much faster and separated into its own faster tick function

CL16gtgh, 2024-02-11:
I extracted read_mcu_status out of this function. Debating but honestly don't think an additional tick function for the interface alone is needed. You can take a look at tick_all_interfaces() in main and let me know if it's good enough.

walkermburns, 2024-02-11:
While that does work, I would like to keep the same naming scheme. My view of it is that only tick_*() functions should be in tick_all. We should also add a suffix to every tick function with their Hz. That way changing the rates of functions is a deliberate change that we have to think about before just moving tick functions around to prevent unintentionally messing with the CANbus load.

closed 2024-04-07 by RCMast3r