hytech-racing/MCU #61: make communication in and out of torque controllers better

RCMast3r, 2024-04-07:


RCMast3r, 2024-04-07:
we can just make the torque controllers each different "systems" with each inheriting from a base class that just has the output function and an input that contains a super-state of the car that has every variable that is needed by all the controllers that gets fed by the sensors of the car. this state estimator should handle all of the filtering and sensor data unification for ease of distributing to the controllers. It can just be given access to the CAN interface and the other sensor interfaces.

walkermburns, 2024-04-08:
This is probably out of scope for FS Invitational, but we can try.
TC MUX and State Estimator should be seperate issues and are each a pretty big project. TC MUX rework is probably higher priority

walkermburns, 2024-04-11:
Taking this out of the Formula South Milestone, we are just going to have to work with what we got for now.

PR #86 (2024-09-20): Feature/tc mux simplification
closes #61
closed 2024-09-20 by jhwang04